aed3/PS4-esp32 #2: DS4 not pairing

dog7431, 2020-02-19:
Hello! Such a problem, I have a LOLIN32 Lite board, I downloaded the sketch and configured the Mac address on dualshock 4, but it does not want to connect. What could be the matter?

tallot13, 2020-03-18:
+1, only blinking led

dmagyar, 2020-03-22:
Strange, it pairs nicely with my android tablet and even if I clone the tablets' BT Mac with the ESP32 it never connects just "blinks". I tried SixAxisPairTool to set a MAC (success) but it still is not pairing with the ESP32... Any pointers?

tallot13, 2020-03-23:
after reset pair, not paired again.. connect on 1 second and reset(and blink)

tallot13, 2020-03-23:
i dont know how :) but it help me
```
pip install esptool
esptool.py --chip esp32 erase_flash
```
after reset, upload again and works

rickwinter2, 2020-03-26:
tallot13 how would i do this please. total noob at esp32. would this be in command prompt or another software? always willing to learn something new. thanks in advance. 
I have had the ps4 controller connect everytime then it just stopped working.

dmagyar, 2020-03-28:
I tried the flash_erase as well as different type of ESP32s. No dice. I can read/set teh MAC with SixAxisTool, I've set it to the example 03:03:03:03:03:03 compile the example - nothing. The middle led blinks on the controller when I turn it on. Is there a special key combination on the controller to force it to pair? This is driving me nuts :(

DerfJagged, 2020-03-28:
It should just connect, but hold the Share and Home buttons until it flashes differently. That's sync mode.

dmagyar, 2020-03-28:
Now the led blinks faster but still no connection to my ESP32. It became visible for my android tablet if I turn on its' Bluetooth. This is my sketch (pretty much sample) compiled with latest platformio...

```
#include <Arduino.h>
#include <PS4Controller.h>

void setup()
{
    Serial.begin(115200);
    PS4.begin("03:03:03:03:03:03");
    Serial.println("Ready.");
}

void loop()
{
                // Below has all accessible outputs from the controller
    if(PS4.isConnected()) {
                        if ( PS4.data.button.up )
                                        Serial.println("Up Button");
                        if ( PS4.data.button.down )
                                        Serial.println("Down Button");
                        if ( PS4.data.button.left )
                                        Serial.println("Left Button");
                        if ( PS4.data.button.right )
                                        Serial.println("Right Button");

                        if ( PS4.data.button.upright )
                                        Serial.println("Up Right");
                        if ( PS4.data.button.upleft )
                                        Serial.println("Up Left");
                        if ( PS4.data.button.downleft )
                                        Serial.println("Down Left");
                        if ( PS4.data.button.downright )
                                        Serial.println("Down Right");

                        if ( PS4.data.button.triangle )
                                        Serial.println("Triangle Button");
                        if ( PS4.data.button.circle )
                                        Serial.println("Circle Button");
                        if ( PS4.data.button.cross )
                                        Serial.println("Cross Button");
                        if ( PS4.data.button.square )
                                        Serial.println("Square Button");

                        if ( PS4.data.button.l1 )
                                        Serial.println("l1 Button");
                        if ( PS4.data.button.r1 )
                                        Serial.println("r1 Button");

                        if ( PS4.data.button.l3 )
                                        Serial.println("l3 Button");
                        if ( PS4.data.button.r3 )
                                        Serial.println("r3 Button");

                        if ( PS4.data.button.share )
                                        Serial.println("Share Button");
                        if ( PS4.data.button.options )
                                        Serial.println("Options Button");

                        if ( PS4.data.button.ps )
                                        Serial.println("PS Button");
                        if ( PS4.data.button.touchpad )
                                        Serial.println("Touch Pad Button");

                        if ( PS4.data.button.l2 ) {
                                        Serial.print("l2 button at ");
                                        Serial.println(PS4.data.analog.button.l2, DEC);
                        }
                        if ( PS4.data.button.r2 ) {
                                        Serial.print("r2 button at ");
                                        Serial.println(PS4.data.analog.button.r2, DEC);
                        }

                        if ( PS4.event.analog_move.stick.lx ) {
                                        Serial.print("Left Stick x at ");
                                        Serial.println(PS4.data.analog.stick.lx, DEC);
                        }
                        if ( PS4.event.analog_move.stick.ly ) {
                                        Serial.print("Left Stick y at ");
                                        Serial.println(PS4.data.analog.stick.ly, DEC);
                        }
                        if ( PS4.event.analog_move.stick.rx ) {
                                        Serial.print("Right Stick x at ");
                                        Serial.println(PS4.data.analog.stick.rx, DEC);
                        }
                        if ( PS4.event.analog_move.stick.ry ) {
                                        Serial.print("Right Stick y at ");
                                        Serial.println(PS4.data.analog.stick.ry, DEC);
                        }

     if (PS4.data.status.charging)
        Serial.println("The controller is charging");
     if (PS4.data.status.audio)
        Serial.println("The controller has headphones attached");
     if (PS4.data.status.mic)
        Serial.println("The controller has a mic attached");

     Serial.print("Battey Percent : ");
     Serial.println(PS4.data.status.battery, DEC);

                 Serial.println();
                 // This delay is to make the Serial Print more human readable
                 // Remove it when you're not trying to see the output
     delay(1000);
    }
}
```

It prints Ready... and then nothing else. Do I have to enable bluetooth on the ESP32 first?

rickwinter2, 2020-03-28:
Got mine working again. Didnt need to flash it. What board are you using dmagyar?

dmagyar, 2020-03-28:
I have many, from TT-GO (with a screen) to ESP32-DEV1 and many barebones. I'm suspecting that the issue is with the controller. It works happily with my Android tablet.

DerfJagged, 2020-03-28:
Try starting sync mode on the controller and then press the reset button on your esp32

dmagyar, 2020-03-28:
Thanks for helping me! Unfortunately this is no dice either. If just prints ready, the controller is in sync mode...

```

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0018,len:4

load:0x3fff001c,len:1044

load:0x40078000,len:8896

load:0x40080400,len:5828

entry 0x400806ac

Ready.
```

tallot13, 2020-03-29:
periodically i have in connect time - short blink, then long and off, connected and reset connect without connected in monitor, in this case helped reset for me

tallot13, 2020-03-29:
> Try starting sync mode on the controller and then press the reset button on your esp32
not helped me

tallot13, 2020-03-29:
@rickwinter2 yes it command promt. install pip and run this commands

tallot13, 2020-03-30:
@rickwinter2 same fix https://github.com/NURobotics/PS4-esp32/issues/4

rickwinter2, 2020-03-30:
I dont know about anyone else but I get different reliability on this library depending on the board I'm using. Found a esp-wroom-32 dev module v3 is the most stable and connects every time. Yet a d1 mini pro and others just keep dropping connection.

keiramayfuckingbe, 2020-05-09:
I have a similar problem. I have a silicon lab's board. It connected easily beforehand, now it won't even print the ready! at the start unless I press the RST and still won't connect.

I did try uninstalling/reinstalling the whole process maybe I put the libraries incorrectly or accidentally pressed reset when I shouldn't have. It will run normal Void loop serialprints fine just as a test? This noob is overloaded.

rickwinter2, 2020-05-09:
Lol got it working now. Bit of a trial but now have a process. 
Using esp32

Have to erase flash.
Write then the mac address to controller
Then rewrite to your board

rickwinter2, 2020-05-09:
I swap controllers between the ps4 and esp32 so have to do this everytime i go back to the esp32
Have you worked out the erase flash process?

ericm65536, 2020-10-02:
I had the same connection problems.  Initially everything worked perfectly, no problems connecting.  Then after a while, the controller would not connect anymore.  But the suggestion from tallot13 using esptool.py somehow fixed the problem

wasista, 2021-01-30:
first: try to uses Ps3Controller.cpp and pair with PS3/PS4. after connected, reset ESP32 and...
the second:
Try to change definition of extern PS4Controller PS4; become PS4Controller Ps4; (lower case 's')
in the end line of two files PS4Controller.cpp and PS4Controller.h
and then edit your program with Ps4.begin("PS4 MAC Addr after set with SixPair"); (lower case 's')
after ESP32 reset, and program running, press Home button only on PS4 controller.
have nice pair, ok ^_^

the-red-eye-studio, 2021-08-07:
> i dont know how :) but it help me
> ```
> pip install esptool
> esptool.py --chip esp32 erase_flash
> ```
> after reset, upload again and works

Thanks man! Worked like a charm, back to work on a ds4 controled football table XD

MarioMarkov, 2023-03-16:
Do this :
```
pip install esptool
esptool.py --chip esp32 erase_flash
```
(here it will show you esp mac)
```
Serial port COM6
Connecting........____
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: **cc:db:a7:02:d4:b8**
Uploading stub...
```

Restart PC 
Set master for Controller to be the mac adress of the esp (with SixAxisPairTool)
In the code set in the `PS4.begin("esp_mac_adress");`