aed3/PS4-esp32 #4: Different controller issue

xman4242, 2020-03-12:
Hello all! 
When using a Heltec Wifi Kit 32 and the VS Code IDE, I am running into some issues. If I try to use a different PS4 remote (even if it has the same address), it wont pair with the ESP32. In addition, the same remote doesn't want to work with multiple ESP32's. The light will go solid, indicating pairing for a split second, then stop. 
Thanks,
Xavier

danieladugyan, 2020-03-15:
Please let me know if you find a solution. I'm having the exact same problem.

xman4242, 2020-03-15:
I will do that. Nothing found yet, I'm unsure where to begin.

xman4242, 2020-03-17:
The solution was found thanks to https://www.reddit.com/r/esp32/comments/fijr4a/bluetooth_debugging/?utm_medium=android_app&utm_source=share. 
Essentially, you have to erase the flash of the esp32 before changing the Mac address. If you use platform io, run: 
pio run -t erase
to erase the flash without the esp-idf.

closed 2020-03-18 by xman4242