aed3/PS4-esp32 #9: PS4 disconnecting a second after connecting

Quantum99k, 2020-08-27:
I'm using the wemos lolin32 and just trying out the code. At first it was working perfectly however after i re uploaded the code for testing it wouldn't keep a connection. The only modification I did was capitalize all "PS4" definition; because some was "Ps4" or "ps4" made it easy to use in place of the PS3esp32 code.

ivo-mallee, 2020-11-25:
i  had this problem too you can fix it by completly erasing the esp32's flash memory with a the esp32 flash tool here is a video about it https://www.youtube.com/watch?v=EGgqdD4oU0U&t=59s. just press the big erase button and wait till it says finished

aed3, 2021-02-09:
I'm going to test out that solution and it to the readme if it does fix the issue. Thanks for bringing this up.

closed 2021-02-09 by aed3