aed3/PS4-esp32 #10: error: format '%x' expects argument of type 'unsigned int*', but argument has type 'uint8_t*

Yeok-c, 2020-10-12:
I cant get it to compile for my ESP32 :/

I'm getting errors (attached below) even while compiling the example program, I think it's because each "byte" of the MAC address is declared as unsigned int in esp_bt_defs.h and again as unsigned char in PS4Controller.cpp.

In esp_bt_defs.h: #define ESP_BD_ADDR_STR "%02x:%02x:%02x:%02x:%02x:%02x"

In PS4Controller.cpp: #define ESP_BD_ADDR_HEX_PTR(addr) (uint8_t*)addr+0, (uint8_t*)addr+1, (uint8_t*)addr+2, (uint8_t*)addr+3, (uint8_t*)addr+4, (uint8_t*)addr+5

Please tell me if I should attach anything else that my may help

Warnings during compilation:
C:\Arduino\libraries\PS4-esp32-master\src\PS4Controller.cpp: In member function 'bool PS4Controller::begin(char*)':
C:\Arduino\libraries\PS4-esp32-master\src\PS4Controller.cpp:50:60: error: format '%x' expects argument of type 'unsigned int*', but argument 3 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]
if (sscanf(mac, ESP_BD_ADDR_STR, ESP_BD_ADDR_HEX_PTR(addr)) != ESP_BD_ADDR_LEN){
^
C:\Arduino\libraries\PS4-esp32-master\src\PS4Controller.cpp:50:60: error: format '%x' expects argument of type 'unsigned int*', but argument 4 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]
C:\Arduino\libraries\PS4-esp32-master\src\PS4Controller.cpp:50:60: error: format '%x' expects argument of type 'unsigned int*', but argument 5 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]
C:\Arduino\libraries\PS4-esp32-master\src\PS4Controller.cpp:50:60: error: format '%x' expects argument of type 'unsigned int*', but argument 6 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]
C:\Arduino\libraries\PS4-esp32-master\src\PS4Controller.cpp:50:60: error: format '%x' expects argument of type 'unsigned int*', but argument 7 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]
C:\Arduino\libraries\PS4-esp32-master\src\PS4Controller.cpp:50:60: error: format '%x' expects argument of type 'unsigned int*', but argument 8 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]
C:\Arduino\libraries\PS4-esp32-master\src\PS4Controller.cpp:50:60: error: format '%x' expects argument of type 'unsigned int*', but argument 3 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]
C:\Arduino\libraries\PS4-esp32-master\src\PS4Controller.cpp:50:60: error: format '%x' expects argument of type 'unsigned int*', but argument 4 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]
C:\Arduino\libraries\PS4-esp32-master\src\PS4Controller.cpp:50:60: error: format '%x' expects argument of type 'unsigned int*', but argument 5 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]
C:\Arduino\libraries\PS4-esp32-master\src\PS4Controller.cpp:50:60: error: format '%x' expects argument of type 'unsigned int*', but argument 6 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]
C:\Arduino\libraries\PS4-esp32-master\src\PS4Controller.cpp:50:60: error: format '%x' expects argument of type 'unsigned int*', but argument 7 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]
C:\Arduino\libraries\PS4-esp32-master\src\PS4Controller.cpp:50:60: error: format '%x' expects argument of type 'unsigned int*', but argument 8 has type 'uint8_t* {aka unsigned char*}' [-Werror=format=]
cc1plus.exe: some warnings being treated as error

aed3, 2021-02-09:
I'm in the process of removing this part of the code, so once that's done I'll close the issue. Thanks for bringing it to my attention though.

Skiri, 2021-02-27:
Also in PS4Controller.cpp  line 57 - bool PS4Controller::end() {}

thirstyone, 2023-03-25:
error: format '%x' expects argument of type 'unsigned int *', but argument 3 has type 'uint8_t *' {aka 'unsigned char *'} [-Werror=format=]
  112 |                     sscanf(message + i, "%02x", &byte_seq[i/2]);
      |                                          ~~~^   ~~~~~~~~~~~~~~
      |                                             |   |
      |                                             |   uint8_t * {aka unsigned char *}
      |                                             unsigned int *
      |                                          %02hhx
but that compiles on other platforms..