aed3/PS4-esp32 #11: I'm having problems connecting to my M5StickC device...

jeffmikels, 2020-12-02:
I'm using the code in the examples, but I can't get it to connect to my device.

Steps:
* I connected my controller to my phone.
* I got the BT mac address from my phone.
* I turn off the BT on my phone.
* I call `PS4.begin("phone mac address here")` in my code
* The controller never connects

aed3, 2021-02-09:
For what you're doing wrong, I believe it comes down to one of two reasons. My first guess is the controller you're using as after experimenting with controllers made by Sony and those not, the third party ones are very hit-or-miss when connecting to the ESP32 even when they work just fine with the PS4. The other problem might be related to the issue mentioned here (#9), so I'd recommend trying that solution and seeing if connecting works from there.

The reason why pairing works in that repo is because it's for different hardware. The USB host shield supports the pairing protocols needed to connect it to a controller, while ESP32s support for it is finicky at best. I've been looking into getting pairing working, but it's a major headache. If I do solve it, I'll be sure to add it to this library.

HawkCollective2890, 2021-05-20:
Try rolling back ESP32 Board def to 1.0.4 fixed it for me.