aed3/PS4-esp32 #12: If library is in use, OTA fails most of the time

Markusenz, 2021-01-03:
I encounter a problem with OTA as soon as I have the library in use.
Even when no ps4 controller is connected,  but ps4.begin(..) is in the setup routine, OTA fails most of the time and I have to reboot the ESP32 several times and try again. After several attempts OTA works.
When removing library use, OTA works every attempt.
Is there any known problem with OTA or a parameter I can tune to make it more stable?

aed3, 2021-02-09:
Hi! I'm unfamiliar with what OTA is, could you explain it? Right now I'm working on some major changes to the library to fix up some of the problems people have mentioned and it will work faster once I'm done, which might solve your second issue.

av4625, 2021-11-24:
OTA is Over The Air updating.

I'm not sure how well using WiFi and this Bluetooth PS4 controller library will work together. I think WiFi and Bluetooth use the same antenna on the ESP32?