aed3/PS4-esp32 #13: no compiling from Arduino IDE

welverin, 2021-01-20:
this the error

libraries/PS4-esp32-master/ps4_gap.c.o:(.literal.ps4_gap_init_service+0x1c): undefined reference to `GAP_ConnOpen'
libraries/PS4-esp32-master/ps4_gap.c.o:(.literal.ps4_gap_event_handle+0xc): undefined reference to `GAP_ConnGetL2CAPCid'
libraries/PS4-esp32-master/ps4_gap.c.o:(.literal.ps4_gap_event_handle+0x10): undefined reference to `GAP_ConnBTRead'
libraries/PS4-esp32-master/ps4_gap.c.o:(.literal.ps4_gap_send_hid+0x14): undefined reference to `GAP_ConnBTWrite'
libraries/PS4-esp32-master/ps4_gap.c.o: In function `ps4_gap_init_service':
/Users/leo/Documents/Arduino/libraries/PS4-esp32-master/src/ps4_gap.c:60: undefined reference to `GAP_ConnOpen'
libraries/PS4-esp32-master/ps4_gap.c.o: In function `ps4_gap_update_connected':
/Users/leo/Documents/Arduino/libraries/PS4-esp32-master/src/ps4_gap.c:60: undefined reference to `GAP_ConnGetL2CAPCid'
/Users/leo/Documents/Arduino/libraries/PS4-esp32-master/src/ps4_gap.c:60: undefined reference to `GAP_ConnGetL2CAPCid'
libraries/PS4-esp32-master/ps4_gap.c.o: In function `ps4_gap_event_handle':
/Users/leo/Documents/Arduino/libraries/PS4-esp32-master/src/ps4_gap.c:60: undefined reference to `GAP_ConnBTRead'
libraries/PS4-esp32-master/ps4_gap.c.o: In function `ps4_gap_send_hid':
/Users/leo/Documents/Arduino/libraries/PS4-esp32-master/src/ps4_gap.c:103: undefined reference to `GAP_ConnBTWrite'
collect2: error: ld returned 1 exit status
exit status 1
Errore durante la compilazione per la scheda Node32s.

welverin, 2021-01-20:
resolved to install 1.0.4 instead 1.05rc of esp32 sdk

closed 2021-01-20 by welverin

drunkscientist, 2021-02-19:
> resolved to install 1.0.4 instead 1.05rc of esp32 sdk

what does this mean? im getting these exact errors as above but am using vscode and platformIO and cannot understand why

edleno2, 2021-02-26:
I think I've reproduced this also - for visual studio code, do F1 for the command pallette, then Arduino:Board Manager, then search on ESP32.  I had taken the latest 1.05.  But there is a version dropdown, I selected 1.04, pressed install.  When that completed (kind of long) I went into the .vscode folder and c_cpp_properties.json and changed the version reference to 1.04 and rebuilt.  Got past the errors about L2CAPCid and others.  Also, if you use the "Include library" button from the library manager it picks up three .h files - PS4_int.h, ps4.h and PS4Controller.h.  But that's the wrong order because ps4_int.h is dependent on PS4.h, but doesn't do a check.  BUT, if you just include PS4Controller.h it will pull all of the dependent includes in the right order.

edleno2, 2021-02-26:
@drunkscientist  - forgot to tag you - take a look at my response above this comment.

welverin, 2021-02-26:
> > resolved to install 1.0.4 instead 1.05rc of esp32 sdk
> 
> what does this mean? im getting these exact errors as above but am using vscode and platformIO and cannot understand why

in arduino ide i install espreiff esp32 library version 1.04 instead of 1.05.xx beta, so i compile without error

drunkscientist, 2021-02-27:
@edleno2 BIG thank you, thats a bunch of menus i had not found yet. unfortunately i was apparantly already at 1.04, have been trying different versions but no changes yet. not sure what you changed in the .json file, mine says its an autogenerated thing and to edit platformio.ini instead