aed3/PS4-esp32 #15: Espressif's internal API

Skiri, 2021-02-27:
The library could be compiled with esp32 boards definition version 1.0.5
It's worth to mention this [esp32-ps3' issue](https://github.com/jvpernis/esp32-ps3/issues/28) in readme section.

aed3, 2021-03-06:
Thanks for mentioning that, I'll make note of it and something to the readme about it.

jmarangola, 2021-03-10:
I have used this library in the past, and I was wondering how I can still use this library given its incompatibility with newer versions.

AkbarRamzan, 2021-03-14:
@Skiri Do you know how to get this library working with 1.0.5?

DaveVdE, 2021-03-17:
I'm also interested to learn how we can make this fabulous library compatible with the latest framework, and even added to the official list of Arduino libraries. How can I help?

Skiri, 2021-03-20:
> @Skiri Do you know how to get this library working with 1.0.5?

I mistyped, of course I wanted to type "The library **could not** be compiled with esp32 boards definition version 1.0.5"

bechhansen, 2021-03-23:
I have the same problem and would love to use this library. I'm building a RC model.

jmarangola, 2021-03-23:
Are you using PlatformIO? I have had success getting it to work recently

bechhansen, 2021-04-03:
Any news or do anyone have a workaround? I tried to install an old esp32 boards definition but that did not work on macOS Big Sur. Apparently another bug.

LorenzoRep, 2021-04-18:
> 
> 
> Are you using PlatformIO? I have had success getting it to work recently

I tried with Arduino IDE and moving to the older 1.0.4 did the trick. But I want to use PlatformIO but so far no luck. Can you help me with that?

DaveVdE, 2021-04-18:
Here's mine:

```
[env:esp32dev]
platform = espressif32@3.0.0
board = esp32dev
framework = arduino
lib_deps = 
	waspinator/AccelStepper@^1.61
	https://github.com/aed3/PS4-esp32/archive/master.zip
```

LorenzoRep, 2021-04-19:
> 
> 
> Here's mine:
> 
> ```
> [env:esp32dev]
> platform = espressif32@3.0.0
> board = esp32dev
> framework = arduino
> lib_deps = 
> 	waspinator/AccelStepper@^1.61
> 	https://github.com/aed3/PS4-esp32/archive/master.zip
> ```

Thank you Dave, this worked!

juthomas, 2021-05-04:
> 
> 
> Here's mine:
> 
> ```
> [env:esp32dev]
> platform = espressif32@3.0.0
> board = esp32dev
> framework = arduino
> lib_deps = 
> 	waspinator/AccelStepper@^1.61
> 	https://github.com/aed3/PS4-esp32/archive/master.zip
> ```

Thanks ! It's perfect (Just cpy and rm "waspinator/AccelStepper@^1.61" and that fix it)

Mendrzec, 2021-10-04:
This is solved with https://github.com/aed3/PS4-esp32/pull/25. @aed3 can you close the issue? Thanks!

closed 2021-10-04 by aed3