aed3/PS4-esp32 #16: having progbelm with uploading to ESP32

mirgreen, 2021-03-13:
when i hit upload i get these errors.
 
[image]

AkbarRamzan, 2021-03-14:
I have the same problem
[image]
Remove version 1.0.5 and instead use 1.0.4
Unfortunately for me, 1.0.4 isn't supported on a raspberry pi 64-bit host so I had to use my windows computer to flash it.

Donatelllo, 2021-03-14:
> I have the same problem
> [image]
> Remove version 1.0.5 and instead use 1.0.4
> Unfortunately for me, 1.0.4 isn't supported on a raspberry pi 64-bit host so I had to use my windows computer to flash it.

this solved the problem. I dont know how to thank. You save me lots of time. Thx

AkbarRamzan, 2021-03-14:
No problem but unfortunately I am stuck :(

naldoxable, 2021-03-14:
My board is not supported in the previous version AHAHAH :(

Donatelllo, 2021-03-16:
Don't  worry. Even if your board supports there may be another problem. Like unable to connect Ps4 to Esp32 :) I have been trying everything but no success

shurillu, 2021-03-16:
@Donatelllo have you tried to completely erase the flash? Something like this:

`esptool.py --chip esp32 erase_flash`

This saved me.

Stefano

AkbarRamzan, 2021-03-16:
> Don't worry. Even if your board supports there may be another problem. Like unable to connect Ps4 to Esp32 :) I have been trying everything but no success

Works perfectly fine for me on windows with 1.0.4. And yes, you have to erase flash if you have connection issues
I just need support for 1.0.5 so I can use my pi

DaveVdE, 2021-03-17:
> Don't worry. Even if your board supports there may be another problem. Like unable to connect Ps4 to Esp32 :) I have been trying everything but no success

What have you tried? I was succesful in using my raspberry pi to pair with the controller and then use the pi's mac address in my own sketch.

Donatelllo, 2021-03-17:
> @Donatelllo have you tried to completely erase the flash? Something like this:
> 
> `esptool.py --chip esp32 erase_flash`
> 
> This saved me.
> 
> Stefano

Sorry but I am new at this. I dont know how to do that on windows.

Donatelllo, 2021-03-17:
> > Don't worry. Even if your board supports there may be another problem. Like unable to connect Ps4 to Esp32 :) I have been trying everything but no success
> 
> What have you tried? I was successful in using my raspberry pi to pair with the controller and then use the pi's mac address in my own sketch.

I followed steps with great cautions. Let me basicly explain you what i did. I found my PS4 (CLone) controller's MAC address with SixAxisPairTool. Then i write that MAC addres to an example code . 

#include <PS4Controller.h>
 
void setup()
{
  Serial.begin(115200);
 
  PS4.begin("3C:61:05:29:7F:BE");             
  Serial.println("Initialization ready!");
}
 
void loop()
{
  if(PS4.isConnected()) {
    Serial.println("Controller connected");
  }
 
  delay(1000);
}

I load this to an esp32 devkit board then i hit PS button on the controller. Unfortunatelly nothing happens. Lastly, there is only one led ESP32  flashs.

AkbarRamzan, 2021-03-17:
@Donatelllo 

Method 1: https://youtu.be/EGgqdD4oU0U

Method 2:
Assuming you have pip
Press the windows key to bring up the start menu. Type "command prompt" (without the quotes) and press enter, this will open a black window
Type "pip install esptool" in the window and hit enter. This will install esptool assuming you have pip. It might say it's already installed.
Type "pip show esptool" in the window, it will tell you where it is installed. You can select it with your mouse and right click to paste. In my case, the path was: "c:\users\akbar\appdata\local\packages\pythonsoftwarefoundation.python.3.7_qbz5n2kfra8p0\localcache\local-packages\python37\site-packages"
Type "cd " (with a space) and then right click to paste the path. i.e. "cd c:\users\akbar\appdata\local\packages\pythonsoftwarefoundation.python.3.7_qbz5n2kfra8p0\localcache\local-packages\python37\site-packages"
This will now change your working directory to where esp tool is located
Make sure your esp32 is connected by usb
Type or paste the command: "python esptool.py --chip esp32 erase_flash"

After erase:
You can now try writing the program to the esp32 again and it should hopefully work

DaveVdE, 2021-03-17:
> I followed steps with great cautions. Let me basicly explain you what i did. I found my PS4 (CLone) controller's MAC address with SixAxisPairTool. Then i write that MAC addres to an example code . [...]

OK, I think I see your problem. You entered the PS4's MAC address, but you need to enter the MAC address of the Bluetooth device that the PS4 is paired with. 

I hope that helps.

Donatelllo, 2021-03-17:
> @Donatelllo
> 
> Method 1: https://www.espressif.com/en/support/download/other-tools
> Download flash download tool, extract all, run flash_download_tool.exe, select esp32, click erase.
> 
> Method 2:
> Assuming you have pip
> Press the windows key to bring up the start menu. Type "command prompt" (without the quotes) and press enter, this will open a black window
> Type "pip install esptool" in the window and hit enter. This will install esptool assuming you have pip. It might say it's already installed.
> Type "pip show esptool" in the window, it will tell you where it is installed. You can select it with your mouse and right click to paste. In my case, the path was: "c:\users\akbar\appdata\local\packages\pythonsoftwarefoundation.python.3.7_qbz5n2kfra8p0\localcache\local-packages\python37\site-packages"
> Type "cd " (with a space) and then right click to paste the path. i.e. "cd c:\users\akbar\appdata\local\packages\pythonsoftwarefoundation.python.3.7_qbz5n2kfra8p0\localcache\local-packages\python37\site-packages"
> This will now change your working directory to where esp tool is located
> Make sure your esp32 is connected by usb
> Type or paste the command: "python esptool.py --chip esp32 erase_flash"
> 
> After erase:
> You can now try writing the program to the esp32 again and it should hopefully work

tried both but no luck :(
I'll try with another ps4 controller then post the results here

Donatelllo, 2021-03-17:
> OK, I think I see your problem. You entered the PS4's MAC address, but you need to enter the MAC address of the Bluetooth device that the PS4 is paired with.
>
> I hope that helps.

I tried this but no luck. I think I should add my PS4 controllers Mac Address to the code bc I watch a video :)

This is the Mac addrees of ESP32 board
[image]

DaveVdE, 2021-03-18:
That's the WiFi MAC address of the ESP32, which is not related to Bluetooth.

The MAC address passed in the call to `PS4.begin` is the address that the ESP32 will assume. The controller needs to be paired with a device like a PS4 or a computer before you can begin. After pairing the controller, you want your ESP32 to take on the identity of the machine that the controller will try to connect to.

The `begin` call does not initiate the connection from the ESP32 to the controller. It's the controller trying to connect to the ESP32. The `begin` call is just there to start the Bluetooth stack.

naldoxable, 2021-03-18:
Using a generic board and version 1.0.4 it works fine!

I have used an android app to scan bluetooth devices to find the ps4 mac address, previously paired with the controller.
No need to erase the memory for me (i have also tried to use different mac address)

Donatelllo, 2021-03-18:
So, I will explain everything i did with pictures for the last time. I hope i did something wrong and you  correct me:)

Firstly i Download an app to my android and connet to my PS4 Controller. I noted the mac address.
[image]
I add this Address into my code and compiled. Then I hit the PS4 button and wait for the connection. 

Sorry for this much disturbance.

DaveVdE, 2021-03-18:
> So, I will explain everything i did with pictures for the last time. I hope i did something wrong and you correct me:)
>
> Firstly i Download an app to my android and connet to my PS4 Controller. I [...]

You’re quite welcome. I understand the confusion. I will try to explain what needs to be done as clearly as possible.

Before you can do anything with the ESP32, the controller needs to be paired with a device like a PS4 console or a computer. That’s because there’s no pairing mechanism for the ESP32, at least to my knowledge. So you’ll need either a PS4, or a PC or, like I did, a Raspberry Pi. 

When the controller has been paired with this device, you’ll then need to find out the Bluetooth MAC address of the device that is paired with the controller. You don’t need the address of the controller, but of what it is paired to. This address, when you find it, needs to be put in the `begin` line, because the trick to getting everything working is to convince the controller that the ESP32 is the device that it is paired to. You’ll probably want to turn off the PS4, because the controller may get confused.

So tell me, what device do you have that you can pair the controller to? Do you happen to have a Raspberry Pi, for instance? Because then it’s just a matter of showing you the command line stuff needed to get the controller paired and work from there.

Donatelllo, 2021-03-18:
I have just a PC. I think I will pair the controller with my pc then get my PC's Mac addres and put that in `begin` line?

DaveVdE, 2021-03-18:
Yes, you can try that. 

Then, if you can turn off the Bluetooth of the PC, or take everything out of range to try it out, because I’m afraid that if the PC’s Bluetooth stays on that it may get confused.

naldoxable, 2021-03-18:
You can even pair with your android device!

Donatelllo, 2021-03-21:
I have tried almost everything. No luck:(

Donatelllo, 2021-03-22:
I think the problem is the controller's brand. It is not Sony original. I ll try with original one and post the results here.

AkbarRamzan, 2021-03-22:
Bruh you only tell us now it's a counterfeit? lol

Donatelllo, 2021-03-22:
I am not a richie man :(

Donatelllo, 2021-03-24:
hi again. Original controller solved it. Thx

closed 2021-03-27 by Donatelllo