aed3/PS4-esp32 #17: Compilation exemple problem

titigwada971, 2021-03-19:
// ************     exemple programme **************************
#include <PS4Controller.h>

void setup() {
  Serial.begin(115200); 
  PS4.begin("1C:66:6D:AA:C9:AD");
  Serial.println("Ready.");
}

void loop() {
  if (PS4.isConnected()) {
    Serial.println("Connected!");
  }
  delay(3000);
}

//******************************************************************

Arduino : 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Carte : "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Users\thier\AppData\Local\Temp\arduino_modified_sketch_941313\Ps4Connect.ino: In function 'void setup()':

C:\Users\thier\AppData\Local\Temp\arduino_modified_sketch_941313\Ps4Connect.ino:9:32: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

   PS4.begin("1C:66:6D:AA:C9:AD");

                                ^

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src\ps4.c: In function 'ps4Enable':

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src\ps4.c:76:3: warning: implicit declaration of function 'memcpy' [-Wimplicit-function-declaration]

   memcpy(hidCommand.data, hid_cmd_payload_ps4_enable, length);

   ^

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src\ps4.c:76:3: warning: incompatible implicit declaration of built-in function 'memcpy'

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src\ps4.c:76:3: note: include '<string.h>' or provide a declaration of 'memcpy'

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src\ps4.c: In function 'ps4SetBluetoothMacAddress':

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src\ps4.c:219:3: warning: incompatible implicit declaration of built-in function 'memcpy'

   memcpy(baseMac, mac, 6);

   ^

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src\ps4.c:219:3: note: include '<string.h>' or provide a declaration of 'memcpy'

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src\ps4_gap.c: In function 'gapSendHid':

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src\ps4_gap.c:86:3: warning: implicit declaration of function 'memcpy' [-Wimplicit-function-declaration]

   memcpy((uint8_t*)(buffer + 1) + buffer->offset, (uint8_t*)hidCommand, buffer->length);

   ^

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src\ps4_gap.c:86:3: warning: incompatible implicit declaration of built-in function 'memcpy'

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src\ps4_gap.c:86:3: note: include '<string.h>' or provide a declaration of 'memcpy'

libraries\PS4-esp32-master\ps4_gap.c.o:(.literal.gapInitService+0x1c): undefined reference to `GAP_ConnOpen'

libraries\PS4-esp32-master\ps4_gap.c.o:(.literal.gapEventHandle+0xc): undefined reference to `GAP_ConnGetL2CAPCid'

libraries\PS4-esp32-master\ps4_gap.c.o:(.literal.gapEventHandle+0x10): undefined reference to `GAP_ConnBTRead'

libraries\PS4-esp32-master\ps4_gap.c.o:(.literal.gapSendHid+0x14): undefined reference to `GAP_ConnBTWrite'

libraries\PS4-esp32-master\ps4_gap.c.o: In function `gapInitService':

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src/ps4_gap.c:113: undefined reference to `GAP_ConnOpen'

libraries\PS4-esp32-master\ps4_gap.c.o: In function `gapUpdateConnected':

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src/ps4_gap.c:176: undefined reference to `GAP_ConnGetL2CAPCid'

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src/ps4_gap.c:176: undefined reference to `GAP_ConnGetL2CAPCid'

libraries\PS4-esp32-master\ps4_gap.c.o: In function `gapEventHandle':

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src/ps4_gap.c:151: undefined reference to `GAP_ConnBTRead'

libraries\PS4-esp32-master\ps4_gap.c.o: In function `gapSendHid':

C:\Users\thier\Documents\Arduino\libraries\PS4-esp32-master\src/ps4_gap.c:88: undefined reference to `GAP_ConnBTWrite'

collect2.exe: error: ld returned 1 exit status

exit status 1

Erreur de compilation pour la carte ESP32 Dev Module

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

DaveVdE, 2021-03-19:
Hi, this is a known issue. 

Currently the library is incompatible with the Arduino framework 1.0.5. You'll need to downgrade.

See #15

Skiri, 2021-03-20:
1.  Espresif library should be 1.0.4
2. [Check this my answer](https://github.com/aed3/PS4-esp32/issues/10#issuecomment-787066334)
3. include '<string.h>'

titigwada971, 2021-03-29:
Hello Skiri and thank you for your answer.Think about upgrading your
library and samples so that it works with the latest Expressif updates
  Thierry

AhmadAkbari82, 2021-10-04:
close on #25

Mendrzec, 2021-10-04:
As @AhmadAkbari82 mentioned this was already solved. @aed3 can you close the issue?

closed 2021-10-04 by aed3