aed3/PS4-esp32 #19: Does not work with new PS4 generation.

AhmadAkbari82, 2021-03-26:
I do not know exactly where the problem comes from. But the Sixaxispair tool seems to works and I can see the MAC address changing. However, it does not connect to ESP32.
I think it's because of Generation. my PS4 Controller is second generation with the CUH-ZCT2E(or 4-597-175-01 serial number) ​model on it. Can anyone help

DaveVdE, 2021-03-26:
I have the same controller (a red one) and it works for me.

AhmadAkbari82, 2021-03-26:
I tested it with different controllers, but it only works with an older Ps4 Controller. I also tested several different ESP32s.
I do not know why! Maybe the problem is with Sixaxispairtool. Which version do you use?
And if you can try this:
1) Pair the controller with your computer (or mobile phone).
2) Turn off Bluetooth.
3) Connect the controller to the computer using USB.
4) Re-enter your PC (or mobile) MAC address using SixAxisPairtools. (Re-enter the displayed Mac and click the Update button)
5) Disconnect the controller from USB and turn on your computer (or mobile phone) Bluetooth.
6) Press the PS button on your controller.
Can it be connected and paired?

DaveVdE, 2021-03-27:
OK, 

My PC does not have Bluetooth, so I'll be using my Raspberry Pi 4B to pair and my PC to use the SixAxisPairTool program to set the host mac address.

Here's what I did:
* Pair the PS4 controller to the RaspBerry Pi:

```
[bluetooth]# scan on
Discovery started
[CHG] Controller DC:A6:32:EA:3B:2D Discovering: yes
...
[NEW] Device 84:30:95:49:61:29 Wireless Controller
[bluetooth]# pair 84:30:95:49:61:29
Attempting to pair with 84:30:95:49:61:29
[CHG] Device 84:30:95:49:61:29 Connected: yes
[CHG] Device 84:30:95:49:61:29 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device 84:30:95:49:61:29 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 84:30:95:49:61:29 ServicesResolved: yes
[CHG] Device 84:30:95:49:61:29 Paired: yes
Pairing successful
...
[bluetooth]# trust 84:30:95:49:61:29
[CHG] Device 84:30:95:49:61:29 Trusted: yes
Changing 84:30:95:49:61:29 trust succeeded
[CHG] Device 84:30:95:49:61:29 Connected: yes
```

Now the controller stays connected. Then I powered down Bluetooth:
```
[Wireless Controller]# power off
[CHG] Device 84:30:95:49:61:29 Connected: no
Changing power off succeeded
[CHG] Controller DC:A6:32:EA:3B:2D Powered: no
[CHG] Controller DC:A6:32:EA:3B:2D Discovering: no
```

Then I connected the controller to my PC using USB. Using the SixAxisPairTool program, I see the following:

[image]

I pasted the same MAC address in the edit box and clicked Update. There's no indication that anything happened but I think that's an oversight in the program. To be sure, I disconnect the controller and connect again to verify. It shows the same MAC address as before, with my edit still in the edit box:

[image]

Now I disconnect the controller from my computer and turn the Bluetooth back on on the Raspberry PI. 

What happens now is that the controller connects briefly but then disconnects quickly. It no longer wants to connect.

DaveVdE, 2021-03-27:
At this point, I had to hard-reset the controller using a pin and holding the reset button for a few seconds and then pair it with the Raspberry Pi again. 

I suggest you don't use the SixAxisPairTool with the V2 controller.

DaveVdE, 2021-03-27:
Also, after being able to succesfully pair it with my RPi, I could no longer connect to my ESP32. I had to erase the flash (see my other post) to fix this. 

Is there something stored in the Flash upon connecting to the controller?

AhmadAkbari82, 2021-03-27:
I have a similar experience to you.
Maybe in 2nd generation something like BT name or something else, store in controller. anyway I think someone must write another tools like sixaxispairtools for DS4.

Omar-Abdul-Hadi, 2021-05-16:
Hi, I have a similar problem here, I cant seem to find your post on erasing the flash, how can I go about fixing this ? 
my robot was working just fine until I used my controller on my laptop, then it just stopped connecting, I tried using sixaxis tools to change the mac address and even setting it back to my phone (which was what I did originally when it was working) but nothing seems to work.

DaveVdE, 2021-05-16:
You'll need to use the esptool, specifically the following command:
```
esptool.py erase_flash
```

This should erase the part of the flash that contains the information that the bluetooth stack uses to connect with a known device.

Omar-Abdul-Hadi, 2021-05-17:
Thank you ! I ended up finding an example code that scans the flash and gives you the option to erase it. when I read your comment I thought you meant erasing the flash on the PS4 controller. for those curious the example is called bt_remove_paired_devices, run it to check if anything is paired and then change the REMOVE_BONDED_DEVICES variable from 1 to 0, this should clear the flash and fix the issue.

JavadRah, 2021-05-18:
Hi Omar!
I did not understand exactly how your problem was solved. Where do you run bt_remove_paired_devices? Please explain in this regard.

vincentj87, 2021-05-25:
Hi i tried to connect to the esp32 with ps4 stick second gen but it’s not connected i tried ps4 with ith ds4 for pc controller and it succed wirelessly but it showed me different mac address for the bluetooth

dokotheo, 2021-05-28:
Hi, 
I think many of us have same problem recently,
How can I recognize between v1/1st Gen and v2/2nd Gen of PS4 controller ?

or, any of you have connect it successfully?

vincentj87, 2021-05-29:
@dokotheo the second gen ps4 controller has a transparant part in the touchpad so you can see the light shine through it

vincentj87, 2021-05-29:
@DaveVdE are you sure , you are using the second gen not the first gen because in my case it didn't connect at all even after i change the mac adresses

DaveVdE, 2021-05-29:
yes, I’m absolutely sure my controller is a genuine 2nd gen PS4 controller.

vincentj87, 2021-05-29:
@DaveVdE do you have any idea why it didn't connected with controller, did i miss any important step ?

dokotheo, 2021-05-30:
> yes, I’m absolutely sure my controller is a genuine 2nd gen PS4 controller.
> […](#)

can you try to install 'PS4 Controller Tester' in your Android,
and check whether you are using gen1 or gen2 ?

DaveVdE, 2021-05-30:
> @DaveVdE
> 
> 
> 
> can you try to install 'PS4 Controller Tester' in your Android,
> 
> and check whether you are using gen1 or gen2 ?

No, I will not. I've checked the model number on my controller and it matches the 2nd generation. I also purchased it new this year. I don't have an Android device anyway.

JavadRah, 2021-05-31:
> Hi,
> I think many of us have same problem recently,
> How can I recognize between v1/1st Gen and v2/2nd Gen of PS4 controller ?
> 
> or, any of you have connect it successfully?

you can check it easily by reading the Model number: CUZ-ZCT2E for 2nd generation and CUZ-ZCT1E for 1st generation.

dokotheo, 2021-05-31:
mine is 2nd Gen for sure

Gura73, 2021-06-08:
Hello. I wanted to connect a PS4 joystick using your libraries. The joystick is new and not original. It failed to connect. I took the original joystick from a friend and he easily connected. I found information on the net that this is a known problem. I also found one description of one of the solutions.
description is [here](https://www.b4x.com/android/forum/threads/esp32-and-bluetooth.115218/post-803817/) May I ask you to fix something in the libraries so that I can use my joystick. I can't do it myself.
Thanks in advance!

JavadRah, 2021-06-08:
> Hello. I wanted to connect a PS4 joystick using your libraries. The joystick is new and not original. It failed to connect. I took the original joystick from a friend and he easily connected. I [...]

I think you must connect your PS4 controller to a PS4 and update it.

Gura73, 2021-06-09:
> > 
> 
> I think you must connect your PS4 controller to a PS4 and update it.

Thanks, I'll try. But I'm not really sure if this will solve the problem, because the gamepad, as I said, is not original.

sbollaerts, 2021-06-25:
> Hello. I wanted to connect a PS4 joystick using your libraries. The joystick is new and not original. It failed to connect. I took the original joystick from a friend and he easily connected. I [...]

Hello I look at the article you mentioned: it seems to be a workaround applied on the ESP-IDF framework, not the PS4-ESP32 library.

BenMartens15, 2021-07-01:
Hey guys, 
I am also experiencing the issue of not being able to connect my 2nd-gen Dualshock 4 controller to my ESP32 using this library. I have tried connecting my controller to my computer and using my computer's Bluetooth MAC in the code for the ESP32 (with computer Bluetooth off); and I have tried writing the ESP32's Bluetooth MAC address to the controller using SixAxisPairTool. But  the controller would not pair to my ESP32 using either method. In addition, I also tried erasing the ESP32 flash and hard resetting the DS4 controller by pressing the little button on the back, but the controller still wouldn't pair. Has anyone figured out a different solution or have a guess at what the problem might be? 
Thanks in advance.

Gura73, 2021-07-02:
Hello Ben. I still haven't solved this problem.
Clearing the ESP 32 flash memory did not help me either.
I think that something needs to be changed in the PS4-ESP32 library,
based on an issue found in the ESP-IDF framework.
But unfortunately the author of the PS4-ESP32 library does not say anything about this.

ArduinoUno17, 2021-08-02:
Okay, so I've spent pretty much my whole day trying to get this thing to work. I have been on several websites, tried the various solutions (there is one I have yet to try, but I have to figure out how to do it), and I tried to modify the library. I was able to get the Playstation Dualshock 4 [Model: CUH-ZCT2U] to connect to the ESP32 through the ESP_HID_HOST example (https://github.com/espressif/esp-idf/blob/release/v4.2/examples/bluetooth/esp_hid_host/main/esp_hid_host_main.c) I ran it in Eclipse. Again, I was able to connect to the PS4 Dualshock 4 Controller. So, I agree that the issue probably is with the library, or it has something to do with the sdkconfig.h file. I'm not too sure. This is all new to me.

Here is what the Terminal spit out from the ESP_HID_HOST example.
```C
I (445) cpu_start: App cpu up.
I (474) cpu_start: Pro cpu start user code
I (474) cpu_start: cpu freq: 160000000
I (475) cpu_start: Application information:
I (479) cpu_start: Project name:     esp_hid_host
I (484) cpu_start: App version:      1
I (489) cpu_start: Compile time:     Aug  1 2021 18:02:44
I (495) cpu_start: ELF file SHA256:  908482454360755b...
I (501) cpu_start: ESP-IDF:          v4.3-dirty
I (506) heap_init: Initializing. RAM available for dynamic allocation:
I (513) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (519) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (525) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (532) heap_init: At 3FFC75D0 len 00018A30 (98 KiB): DRAM
I (538) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (544) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (551) heap_init: At 40096D0C len 000092F4 (36 KiB): IRAM
I (558) spi_flash: detected chip: generic
I (561) spi_flash: flash io: dio
W (565) spi_flash: Detected size(4096k) larger than the size in the binary image
 header(2048k). Using the size in the binary image header.
I (580) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (610) BTDM_INIT: BT controller compile version [1342a48]
I (610) system_api: Base MAC address is not set
I (610) system_api: read default base MAC address from EFUSE
I (620) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1340) ESP_HIDH_DEMO: SCAN...
I (6350) ESP_HIDH_DEMO: SCAN: 0 results
```

This is the point when I hit the Playstation Button on the controller to turn it on, it connects and this is the data that follows.
```C
E (100440) BT_BTM: tBTM_SEC_DEV:0x3ffd2da4 rs_disc_pending=0

W (100450) BT_SDP: process_service_attr_rsp

E (100460) BT_APPL: wrong device handle: [0]
E (100470) BT_APPL: wrong device handle: [0]
W (100590) BT_SDP: process_service_attr_rsp

I (100620) BT_HIDH: Device Not Found? It's probably a reconnect.
I (100630) ESP_HIDH_DEMO: 98:b6:e9:6b:f2:fa INPUT:  GAMEPAD, MAP:  0, ID:   1, L
en: 9, Data:
I (100630) ESP_HIDH_DEMO: 00 80 80 80 08 fc 36 fd 3f 
I (100630) ESP_HIDH_DEMO: 98:b6:e9:6b:f2:fa INPUT:  GAMEPAD, MAP:  0, ID:   1, L
en: 9, Data:
I (100640) ESP_HIDH_DEMO: 00 80 80 80 08 78 59 fd 3f 
I (100650) ESP_HIDH_DEMO: 98:b6:e9:6b:f2:fa INPUT:  GAMEPAD, MAP:  0, ID:   1, L
en: 9, Data:
I (100660) ESP_HIDH_DEMO: 80 80 80 80 08 70 5c fd 3f 
I (100660) ESP_HIDH_DEMO: 98:b6:e9:6b:f2:fa INPUT:  GAMEPAD, MAP:  0, ID:   1, L
en: 9, Data:
I (100670) ESP_HIDH_DEMO: 80 80 80 80 08 70 5c fd 3f 
I (100680) ESP_HIDH_DEMO: 98:b6:e9:6b:f2:fa OPEN: 
BDA:98:b6:e9:6b:f2:fa, Status: OK, Connected: YES, Handle: 0, Usage: GAMEPAD
Name: , Manufacturer: , Serial Number:
PID: 0x09cc, VID: 0x054c, VERSION: 0x0100
Report Map Length: 430
    VENDOR FEATURE REPORT, ID: 180, Length:  63
    VENDOR FEATURE REPORT, ID: 179, Length:  63
    VENDOR FEATURE REPORT, ID: 178, Length:  63
    VENDOR FEATURE REPORT, ID: 177, Length:  63
    VENDOR FEATURE REPORT, ID: 173, Length:  63
    VENDOR FEATURE REPORT, ID: 172, Length:  63
    VENDOR FEATURE REPORT, ID: 171, Length:  63
[... 647 lines cut ...]
I (103900) ESP_HIDH_DEMO: 98:b6:e9:6b:f2:fa INPUT:  GAMEPAD, MAP:  0, ID:   1, L
en: 9, Data:
I (103910) ESP_HIDH_DEMO: 80 80 80 80 08 00 d4 00 00 
I (103920) ESP_HIDH_DEMO: 98:b6:e9:6b:f2:fa INPUT:  GAMEPAD, MAP:  0, ID:   1, L
en: 9, Data:
I (103930) ESP_HIDH_DEMO: 3f c4 4a fd 3f cc 4a fd 3f 
I (103930) ESP_HIDH_DEMO: 98:b6:e9:6b:f2:fa INPUT:  GAMEPAD, MAP:  0, ID:   1, L
en: 9, Data:
I (103940) ESP_HIDH_DEMO: 80 80 80 80 08 00 00 00 00 
I (103950) ESP_HIDH_DEMO: 98:b6:e9:6b:f2:fa INPUT:  GAMEPAD, MAP:  0, ID:   1, L
en: 9, Data:
I (103960) ESP_HIDH_DEMO: 40 00 00 00 00 34 66 fd 3f 
I (103960) ESP_HIDH_DEMO: 98:b6:e9:6b:f2:fa INPUT:  GAMEPAD, MAP:  0, ID:   1, L
en: 9, Data:
I (103970) ESP_HIDH_DEMO: 40 00 00 00 00 6c 5e fd 3f 
I (103980) ESP_HIDH_DEMO: 98:b6:e9:6b:f2:fa INPUT:  GAMEPAD, MAP:  0, ID:   1, L
en: 9, Data:
I (103980) ESP_HIDH_DEMO: 00 50 81 fd 3f fc 36 fd 3f 
I (103990) ESP_HIDH_DEMO: 98:b6:e9:6b:f2:fa INPUT:  GAMEPAD, MAP:  0, ID:   1, L
en: 9, Data:
I (104000) ESP_HIDH_DEMO: 00 2c 51 fd 3f 0c 00 00 00 
I (104010) ESP_HIDH_DEMO: 98:b6:e9:6b:f2:fa INPUT:  GAMEPAD, MAP:  0, ID:   1, L
en: 9, Data:
I (104010) ESP_HIDH_DEMO: 00 f4 4c fd 3f cc 4b fd 3f
```

mjs513, 2021-08-04:
Is there any resolution to this issue - I have 2 new PS4 controllers that can not connect but if I use my older DS4 controller it works without an issue.

ArduinoUno17, 2021-08-04:
The PS4 Dualshock 4 Controller Model: CUH-ZCT1U seems to work without any issues, it connects immediately and I am able to get data. The PS4 Dualshock 4 Controller Model: CUH-ZCT2U does not, no connection, no data, nothing. I'm going to do some research. Hopefully the author is working on a fix.

KurtE, 2021-08-04:
@mjs513 @ArduinoUno17 @aed3 - Right now it appears like it may be a hit or miss with different PS4 controllers and especially with different clones. 

For example I have a Voyee that works well with this library.  But the one I have from Terrios does not.   

And with doing work on trying to support some of these I know that some of these use some different Bluetooth sub-systems for binding.    Do you know if any of these work with other PS4/USB Hosting type setups like USB Host Shield 2... I do believe some of these do work with Linux... 

I probably also need to get a newer official PS4 controller like  CUH-ZCT2U to see what they are now doing. 

Note: I will probably do most of my experimentation on our own setup (Teensy 4.1 using USBHost_t36).   I will monitor here to see if anyone else has made progress on these.

Kurt

ArduinoUno17, 2021-08-04:
@KurtE I'm trying to stay away from using a USB Host Shield with a Bluetooth USB Dongle. It is possible to successfully pair and bind the PS4 Dualshock 4 Model CUH-ZCT2U with the ESP32.

> I was able to get the Playstation Dualshock 4 [Model: CUH-ZCT2U] to connect to the ESP32 through the ESP_HID_HOST example (https://github.com/espressif/esp-idf/blob/release/v4.2/examples/bluetooth/esp_hid_host/main/esp_hid_host_main.c)

But that was using Eclipse and ESP-IDF, not this Arduino Library. So again, it is possible to pair and connect the CUH-ZCT2U. I did notice that the PID for the CUH-ZCT2U is different from the CUH-ZCT1U. I'm not sure if the code is searching for the particular PID or not. I can't find anything so far. I am new to this kind of stuff. I'm still trying to figure out how all of this works.

KurtE, 2021-08-04:
@ArduinoUno17 - My Model CUH-ZCT2U 

Appears to work here,  When I ran the sixpair tool it had to install some new driver... Then I updated the address... 
Then rebuilt a test program and it is seeing it. 

However there are others which do not work and for those again it is useful to know if others have had luck with it on other platforms as a way to know maybe what the differences may be.

ArduinoUno17, 2021-08-04:
@KurtE are you using the USB Host Shield Method?

KurtE, 2021-08-04:
> @KurtE are you using the USB Host Shield Method?
I am not sure what you are asking?  

For the last posting.  I used the PC app SixAxisPair tool to update the pairing information within the PS4 controller.  I then reprogrammed my Sparkfun ESP32 Thing Plus with a modified sketch.  Some of the mods were to make it build, like needing to include certain header files.  like stdio.h to remove compiler errors.  The Sparkfun version appears to still have older libraries so it links.  

USB Host Shield - I only occasionally use to see if/how something works on it.  My main interests are to improve our support in the Teensy library USBHost_t36.   But I am interested here as I know that some other code I am playing with for Hexapods, that others have adapted it to use an ESP32 and this library to talk to an Robotis openCM9.04 board.

Hope that makes sense

ArduinoUno17, 2021-08-04:
@KurtE what model controller do you have working with the Sparkfun ESP32 Thing Plus? Is it the CUH-ZCT1U or CUH-ZCT2U?

KurtE, 2021-08-04:
[image]

ArduinoUno17, 2021-08-04:
@KurtE what ESP32 Board version are you using from the Board Manager?

KurtE, 2021-08-04:
Note: I am using the Sparfun ESP32 thing plus : 
https://www.sparkfun.com/products/15663

ESP32 setup:  I have the generic ESP32 setup, plus the Sparkfun one..  I am building using Sparkfun 

Version... Sparkfun ESP32-boards V1.0.0
But goes to: C:\Users\kurte\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

ArduinoUno17, 2021-08-04:
@KurtE How did you get it to compile? What files did you have to include?

KurtE, 2021-08-04:
Me too, when I was not using the Sparkfun board types.
[image]

I believe when @mjs513 was building for the main ESP32 install he needed to downgrade to probably 1.04

KurtE, 2021-08-04:
Note: my changes to get it to compile are up in the fork/branch:
https://github.com/kurte/PS4-esp32/tree/build_fixes

These had to do with adding #includes plus changes to call to sscanf to not pass in uint8_t pointers but use temporary 32 bit ones to fill in and then copy in the results...

mjs513, 2021-08-04:
Yep.  To get it to compile for a non-Sparkfun ESP32 you will have to use 1.0.4.  Anything above that will cause errors -- the ESP32 arduino folks changed the BT API after that version

joseortiz1080, 2021-09-12:
> 
> 
> You'll need to use the esptool, specifically the following command:
> 
> ```
> esptool.py erase_flash
> ```
> 
> This should erase the part of the flash that contains the information that the bluetooth stack uses to connect with a known device.

Thanks, I used this information and work great, I recomend this tutorial for the flash process: https://www.youtube.com/watch?v=zfvA8ZVVQYY

spilz87, 2021-12-11:
hello

I think I have the same issue with chinese PS4 gamepad
it works well on computer, but now way to connect it with ESP32 :(

nobody find a solution for that ?

playrobotics-alex, 2022-03-29:
@KurtE 
Hi, can you please tell me if this is the controller you have and it works?
https://www.aliexpress.com/item/1005003496011676.html

KurtE, 2022-03-29:
@playrobotics-alex Sorry, it has been a long while, since I have tried out the different PS4 controllers, but off on other distractions. 

I purchased my voyee from Amazon: https://smile.amazon.com/gp/product/B08ZHFM4B5

Is it the same one?  good question, with several of these clones they may have 10 versions.   All with slightly different firmware. 

And as I mentioned it has been many months since I tried it.  And a very long time since I tried it on ESP32

Sorry I can not give a definitive answer.

playrobotics-alex, 2022-03-29:
Thank you for the info, I will try :-)

spilz87, 2022-06-13:
I don't get what you mean, which contrôler ? I don't find it in your link

vandjac, 2022-06-18:
I just bought the controller that @KurtE recommended, and it worked! If anyone else is having this problem, just buy that controller, it's cheap and it works!

spilz87, 2022-06-18:
Do you have a link?

vandjac, 2022-06-18:
Scroll up a little bit. It's in KurtE's comment.

spilz87, 2022-06-18:
Ok, I was clicking on the name so it opens his repository 🤦‍♂️

So actually for me with shipping and VAT it's become too expensive :(

But thanks for sharing it 😁