aed3/PS4-esp32 #36: Old version with old esp32 firmware available anywhere?

aramich100, 2021-11-26:


JavadRah, 2021-12-01:
use this commit: 35f2caa258875c1dfa1ad73901f4e2505dfeeccb with [this link](https://github.com/aed3/PS4-esp32/tree/35f2caa258875c1dfa1ad73901f4e2505dfeeccb)

aramich100, 2021-12-01:
[image]

aramich100, 2021-12-01:
/Users/michaelarabian/Documents/Arduino/libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/src/ps4_spp.c: In function 'sppCallback':
/Users/michaelarabian/Documents/Arduino/libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/src/ps4_spp.c:89:30: error: 'ESP_BT_SCAN_MODE_CONNECTABLE' undeclared (first use in this function)
     esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE);
                              ^
/Users/michaelarabian/Documents/Arduino/libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/src/ps4_spp.c:89:30: note: each undeclared identifier is reported only once for each function it appears in
/Users/michaelarabian/Documents/Arduino/libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/src/ps4_spp.c:89:5: error: too few arguments to function 'esp_bt_gap_set_scan_mode'
     esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE);
     ^
In file included from /Users/michaelarabian/Documents/Arduino/libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/src/ps4_spp.c:4:0:
/Users/michaelarabian/Library/Arduino15/packages/esp32/hardware/esp32/2.0.1/tools/sdk/esp32/include/bt/host/bluedroid/api/include/api/esp_gap_bt_api.h:479:11: note: declared here
 esp_err_t esp_bt_gap_set_scan_mode(esp_bt_connection_mode_t c_mode, esp_bt_discovery_mode_t d_mode);
           ^
/Users/michaelarabian/Documents/Arduino/libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/src/ps4_spp.c: At top level:
cc1: warning: unrecognized command line option '-Wno-frame-address'
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

aramich100, 2021-12-01:
I now got the library to compile, but as soon as i add the PS4.begin("...") i get this : 

Arduino: 1.8.16 (Mac OS X), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/ps4_gap.c.o:(.literal.gapInitService+0x1c): undefined reference to `GAP_ConnOpen'
libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/ps4_gap.c.o:(.literal.gapEventHandle+0xc): undefined reference to `GAP_ConnGetL2CAPCid'
libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/ps4_gap.c.o:(.literal.gapEventHandle+0x10): undefined reference to `GAP_ConnBTRead'
libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/ps4_gap.c.o:(.literal.gapSendHid+0x14): undefined reference to `GAP_ConnBTWrite'
libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/ps4_gap.c.o: In function `gapInitService':
/Users/michaelarabian/Documents/Arduino/libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/src/ps4_gap.c:48: undefined reference to `GAP_ConnOpen'
libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/ps4_gap.c.o: In function `gapUpdateConnected':
/Users/michaelarabian/Documents/Arduino/libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/src/ps4_gap.c:48: undefined reference to `GAP_ConnGetL2CAPCid'
/Users/michaelarabian/Documents/Arduino/libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/src/ps4_gap.c:48: undefined reference to `GAP_ConnGetL2CAPCid'
libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/ps4_gap.c.o: In function `gapEventHandle':
/Users/michaelarabian/Documents/Arduino/libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/src/ps4_gap.c:48: undefined reference to `GAP_ConnBTRead'
libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/ps4_gap.c.o: In function `gapSendHid':
/Users/michaelarabian/Documents/Arduino/libraries/PS4-esp32-35f2caa258875c1dfa1ad73901f4e2505dfeeccb/src/ps4_gap.c:88: undefined reference to `GAP_ConnBTWrite'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

aramich100, 2021-12-04:
After many tried, i finally got it to work! 
I updated my board manager url to this : https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
now im having trouble connecting my controller..