aed3/PS4-esp32 #38: Fake PS4 gamepad don't connect, any idea ?

spilz87, 2021-12-13:
hello,

I don't have PS4, and was looking for a gamepad that was compatible with ESP32 for DIY project.
So I bought fake PS4 gamepad to use them with this livrary.

I read the readme file and used sixaxispairTool to change the MAC address, but gamepad doesn't connect to the ESP32 :(
I just borrow an official PS4 gamepad, I change the MAC address and it connect directly 

the issue, as suggested in other issues #19 and #29 the issue is with fake PS4 gamepad

I don't know what is the difference..

if someone has any idea to find a way to find a workaround 

thanks in advance

JavadRah, 2021-12-14:
Hi
The original PS4 gamepad has a microcontroller in which the program is hidden. We can only guess it through its behavior. The SixAxisPair tool is actually a hack to find the original Sony gamepad connection register. Therefore, this tool may not necessarily work properly on fake gamepads (in my experience, some fake gamepads are programmed by SixAxisPair and work easily. But "not necessarily" all fake gamepads will work with this tool.)
The best solution, I think, is to change the method. That is, instead of using SixAxisPair, put the gamepad in pairing mode (you have to hold down the Share and PS buttons for a few seconds, the gamepad starts blinking. Google it for more information) and then find and pair it with ESP32. Then write your original ESP32 MAC address in the Ps4.begin field. (SixAxisPair will probably show you this address after pairing. Otherwise, when programming ESP32, find the MAC address in the console.)
I do not know how effective and useful this method is, but it can be worth testing anyway.

spilz87, 2021-12-14:
Hi @JavadRah 
Thanks for your reply
Actually I already tried this :
- Connect the fake game pad to my windows computer through Bluetooth (ok)
- check with sixaxispair if MAC address is the MAC address of the my Bluetooth of my PC (yes it is)
- set the MAC address on ps4.begin
- turn off Bluetooth on my PC
- turn on gamepad
-> game pas don't connect to my ESP32 :(

Doing the same with Sony gamepad works like a charme

spilz87, 2021-12-14:
@JavadRah You said "then find and pair it with ESP32"

How do you do that ? 
(I'm using arduino IDE)

I'm not able to get any connection between fake gamepad / ESP32 :(

spilz87, 2021-12-14:
I feel like the library is looking for an ID or something like this that is not the same on original and fake gamepad

spilz87, 2021-12-14:
An other test :
I found an other fake gamepad.
I did the same process :
- Connect it to my PC
- set the MAC address on my ESP32 
- turn of Bluetooth on PC 
-> the gamepad connect correctly to my ESP32 

Then I wanted to change the MAC address with SixaxisPairTool to have a different one.
- change the MAC address on the gamepad and in the ESP32 
-> gamepad connect then turns off, no "connected" printed :(

Then now way to make it works even by trying to do connect it to the PC 

-> SixaxisPairTool changes something more than the MAC address ?? 

No idea for next step 🤷‍♂️

spilz87, 2021-12-15:
> * change the MAC address on the gamepad and in the ESP32
>   -> gamepad connect then turns off, no "connected" printed :(
> 
> Then now way to make it works even by trying to do connect it to the PC
> 
> -> SixaxisPairTool changes something more than the MAC address ??

this part was solved by ereasing flash of the ESP32
`py -m esptool --chip esp32 --port COM7 erase_flash`

vandjac, 2022-06-13:
Did you get it working @spilz87?

spilz87, 2022-06-13:
No I didn't :(

So I build my own gamepad and I will try with small gamepad working with BLE.

lucasalberto01, 2023-03-20:
I'm facing this exact problem :c

planevina, 2023-07-04:
same problem o.o

JavadRah, 2023-07-04:
> @JavadRah You said "then find and pair it with ESP32"
> 
> How do you do that ? (I'm using arduino IDE)
> 
> I'm not able to get any connection between fake gamepad / ESP32 :(

Hi @spilz87 
I don't know how it is possible, but in basic you can find some Bluetooth devices around yourself. if there's a way to advertise the PS4 controller in BT then you can find it with Arduino.