aed3/PS4-esp32 #40: PS4 Controller connects and turns off

FFelge, 2022-01-25:
When the PS4 Controller cannot connect to/find the ESP32, it just blinks and eventually turns off. But if it connects, it blinks white, then glows white and then glows PS4 blue. And I already successfully connected the Controller to the ESP32 and controlled it with it. But then the next day, without making any changes in the code, the Controller won't connect. It blinks white, glows white but instead of turning blue and fully connect, it just turns off. I then later tried it with a different controller, and now, again without any changes, it connects. About a week later, I tried to connect the controller that worked the last time, and again, it doesn't connect. I'm completely clueless what could possibly cause this issue, I've tried changing the MAC adress in the code and on the controller, but nothing helps.

https://user-images.githubusercontent.com/98420099/151040370-acb2be80-8dbf-4025-9944-9e3e15bf21ba.mp4

https://user-images.githubusercontent.com/98420099/151040516-5c6a7074-5254-4a84-9b20-65a813da717c.mp4

Ninagawa123, 2022-01-30:
I had exactly the same symptoms. The script that worked half a month ago has stopped working.

FFelge, 2022-01-30:
> I had exactly the same symptoms. The script that worked half a month ago has stopped working.

Did you find a solution yet? Cause I tried so many things and none of them worked. The same script literally stopped working and then started working again on the same day. Now it’s broken again, frustrating

Ninagawa123, 2022-01-31:
> Did you find a solution yet? Cause I tried so many things and none of them worked. The same script literally stopped working and then started working again on the same day. Now it’s broken again, [...]

Not yet. I have no choice but to use wiimote instead.

FFelge, 2022-01-31:
> Not yet. I have no choice but to use wiimote instead.

Do you mean a Wii remote or is Wiimote a library?

Ninagawa123, 2022-01-31:
> 

I use a wii remote controller　with library for a while. But I want to use a ps4 controller.

chrisdoel, 2022-02-03:
Hi, there is a fix for this issue. It seems to be caused by the paired devices being stored in flash.

You can temporarily fix it by using esptool.py to clear the flash. 

A permanent fix that i found is to copy the following example code which removes the paired devices, and run it at the start of setup (before initialising the ps4 library).

https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/bt_remove_paired_devices/bt_remove_paired_devices.ino

FFelge, 2022-02-03:
> Hi, there is a fix for this issue. It seems to be caused by the paired devices being stored in flash.
>
> You can temporarily fix it by using esptool.py to clear the flash.
>
> A permanent fix that i [...]

Thanks, I‘ll definitely try that! But the code you sent involves also the void setup, so I can’t put the whole code in front of the initialization of the library..
I‘m going to look into the code later, maybe I‘ll find out on my own.

UtzLe, 2022-02-03:
> Hi, there is a fix for this issue. It seems to be caused by the paired devices being stored in flash.
>
> You can temporarily fix it by using esptool.py to clear the flash.
>
> A permanent fix that i [...]

Thanks for that.

Ninagawa123, 2022-02-04:
> Hi, there is a fix for this issue. It seems to be caused by the paired devices being stored in flash.
>
> You can temporarily fix it by using esptool.py to clear the flash.
>
> A permanent fix that i [...]

Thank you so much and it worked well !!

mbarabasz, 2022-02-04:
Thanks @chrisdoel your solution works!

FFelge, 2022-04-22:
> Hi, there is a fix for this issue. It seems to be caused by the paired devices being stored in flash.
>
> You can temporarily fix it by using esptool.py to clear the flash.
>
> A permanent fix that i [...]

I`ve tried it by just copying the part before the void setup in mycode before initialising my ps4 library and copying the part in the void setup into my void setup. Now my controller won't even connect. Do I have the wrong ESP? I got the ESP32 DEVKIT V1.
Or else, could you share your working code? Thank you in advance!

gigahidjrikaaa, 2023-04-18:
> 

I still have a problem when I use that code snippet.

I constantly need to upload the code to the ESP and then reupload the code without the snippet. After only that, the controller is able to connect. If the ESP happens to be power cycled, then I have to repeat this same process once again. Is there any way to solve this?