aed3/PS4-esp32 #41: How can I disconnect the PS4 controller?

vohai2003, 2022-02-12:
Just curious, how can I disconnect the PS4 controller without shutting it down?

changfrancis, 2022-05-01:
I tried the same code with a clone and original PS4 controller. The library does not seems to work on my clone controller. It work perfectly on the original controller. 

The clone on with all correct "mac address", ESP32 erase method or fancy pairing method using PC and transfer the Mac address all does not work correctly. most of the time the clone controller will attempt connection with ESP32 and shutdown (<2sec). I am not a expert programmer to find out what is wrong, why it does not work with the clone version.

I hope my finding helps you guys who wanna continue using PS4 as a controller. 

If anyone find any other library for a reliable controller pls let me know, I will wish to try xbox or switch controller, but i dont see much library out there.

vohai2003, 2022-05-19:
> I tried the same code with a clone and original PS4 controller. The library does not seems to work on my clone controller. It work perfectly on the original controller. 
> 
> Witht the clone, i tried correcting "mac address", ESP32 erase method or fancy pairing method using PC and transfer the Mac address all does not work correctly. most of the time the clone controller will attempt connection with ESP32 and shutdown (<2sec). I am not a expert programmer to find out what is wrong, why it does not work with the clone version.
> 
> I hope my finding helps you guys who wanna continue using PS4 as a controller. 
> 
> If anyone find any other library for a reliable controller pls let me know, I will wish to try xbox or switch controller, but i dont see much library out there.

You may check if your code have any interrupt command. If there's any, it may shut your PS4 down after a several seconds.

Zwer2k, 2022-07-08:
Same question from me. How can I disconnect the controller without restarting ESP?

ullMarksin, 2022-09-06:
if you hold ps button for 5 sec , controller will turn off