aed3/PS4-esp32 #44: Which 3rd party remotes will work with this library?

playrobotics-alex, 2022-03-28:
I know that there are many issues with 3rd party remotes not working with this library but I was wandering if anyone would be able to confirm which brand / model was proven to work? I just want to buy a controller that will work :-)
Thanks

rkuo2000, 2022-04-15:
I just bought a 3rd-party PS4 controller, it can connect to PC, Smartphone, but it can not connect to ESP32 using this library.

vandjac, 2022-06-13:
I am also wondering this. I bought this controller: https://www.amazon.com/gp/product/B097XJRTBR/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

It connects by bluetooth to my PC but not to the ESP32.

Beachtech1, 2022-09-18:
I also tried the controller above. I'm trying to use it in a maker space for sumo robots. It didn't work, but I found a thread on reddit of the original creator of the PS4 library committing to the issue.

https://www.reddit.com/r/esp32/comments/krp6kr/third_party_ps4_controller_and_esp32/

sinoning, 2022-09-27:
why the copy ps4 not work?

rkuo2000, 2022-10-18:
My clone-PS4 controller can not pair with PC/Smartphone,
not sure if it is because Model
The clone-PS4 Model. CUH-ZCT2E (not working)
The original-PS4 Model. CUH-ZCT2G (works well)