aed3/PS4-esp32 #47: Can pair only one time after erase_flash

muek14, 2022-08-15:
I use this controller https://www.amazon.de/gp/product/B09ZKTNS5F/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 have noch problems to change MAC with SixPairTools and can also pair to my ESP32 but it's only possible one time, when I have done the command esptool.py --chip esp32 erase_flash before flashing the program. After a reset of my ESP32 and I try to pair again, the light on the controller ist flashing 2-4 times, but no more pairing ist possible. 

After erase_flash and flashig the program again, pairing is possible.

Have anyone an idea what is wrong in this library?

beniroquai, 2022-09-09:
same here, did you find a solution for this? somehow related. #40?

muek14, 2022-09-11:
> same here, did you find a solution for this? somehow related. #40?

oh yes, thanks for this hint, with this workaround I can life, I habe added the relevant code lines to remove the paired devices in my init function, and now it works fine

best regards from vienna/austria Klaus

closed 2022-09-11 by muek14