aed3/PS4-esp32 #50: Unable to find the PS4 MAC on with the sixaxis pair tool

ypatidisioannis, 2022-11-25:
Hello guys!

I need help. I have bought a fake PS4 controller the ASPIS3 , downloaded the sixaxis pair tool and connect my controller with the USB to my windows PC. 

The tool says it doesn't foundany device. Although my controller works very good with controlling windows.

Any ideas what to do?

P.S 

Already tried and disabled the firewall and antivirus.

ypatidisioannis, 2022-11-25:
Although I have paired my fake controller with my android wireless.

I have put the bluetooth MAC address of the Android to my ESP32 device and still doesn't connect. Which it should , shouldn't?

thebigboss84, 2023-04-05:
We're you able to fix it ? I have the same problem with Axis tool

ypatidisioannis, 2023-04-05:
Hello, 

Yes, the solution is that I am using only SONY PS3 controllers only. All the others are having problems. I wouldn't use the other controllers , only from SONY