aed3/PS4-esp32 #51: Compiling Error with ESP32 and ESP32-S3

Fox82, 2023-02-14:
Hi All,
installed the library with IDE 2.0.3, tested also with 1.8.13 and 1.6.9. 
Here is the following compiler error:

Arduino: 1.6.9 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Users\Fox\Documents\Arduino\libraries\PS4-esp32-master\src\ps4_spp.c: In function 'sppCallback':

C:\Users\Fox\Documents\Arduino\libraries\PS4-esp32-master\src\ps4_spp.c:87:30: error: 'ESP_BT_CONNECTABLE' undeclared (first use in this function)

     esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);

                              ^

C:\Users\Fox\Documents\Arduino\libraries\PS4-esp32-master\src\ps4_spp.c:87:30: note: each undeclared identifier is reported only once for each function it appears in

C:\Users\Fox\Documents\Arduino\libraries\PS4-esp32-master\src\ps4_spp.c:87:50: error: 'ESP_BT_NON_DISCOVERABLE' undeclared (first use in this function)

     esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);

                                                  ^

C:\Users\Fox\Documents\Arduino\libraries\PS4-esp32-master\src\ps4_spp.c:87:5: error: too many arguments to function 'esp_bt_gap_set_scan_mode'

     esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);

     ^

In file included from C:\Users\Fox\Documents\Arduino\libraries\PS4-esp32-master\src\ps4_spp.c:4:0:

C:\Users\Fox\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/bt/esp_gap_bt_api.h:370:11: note: declared here

 esp_err_t esp_bt_gap_set_scan_mode(esp_bt_scan_mode_t mode);

           ^

exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

any ideas how to solve it?

gerokatseros, 2023-02-19:
Same error on me too while compiling the code

mattwilliamson, 2023-02-19:
Just comment out these three commented lines in `ps4_spp.c`:

```c
// #if CONFIG_IDF_COMPATIBILITY >= IDF_COMPATIBILITY_MASTER_D9CE0BB
    // esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);
// #elif CONFIG_IDF_COMPATIBILITY >= IDF_COMPATIBILITY_MASTER_21AF1D7
    esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE);
// #endif
```